Vrekrer/Vrekrer_scpi_parser #5: Projects using Vrekrer_scpi_parser

Vrekrer, 2020-05-23:
If you use Vrekrer_scpi_parser library, please post your project here.

Vrekrer, 2020-05-23:
Vrekrer_Voltmeter
14 bit voltimeter with SCPI usb and ethernet interfaces for Arduino
https://github.com/Vrekrer/Vrekrer_Voltmeter

Vrekrer, 2020-05-23:
Vrekrer_Step_Motor
SCPI step motor controler for Arduino
https://github.com/Vrekrer/Vrekrer_Step_Motor

codenamezoo, 2020-05-26:
https://github.com/codenamezoo/Vrekrer_scpi_parser

rselec, 2020-12-08:
Hi,
I'm going to use it for my TinyFanController to implement USB communication.
Things are working in general but I have some questions/issues which are also something for doku I think.

1. What are the define SCPI_MAX_TOKENS for, how could I count how many Tokens I have and what is meant with token?
2. Where should I overwrite the configuration defines? Defining them in main.cpp gives redefined warnings and won't work and modifying the lib header is not a way, because it is outside the application code.

Many Thanks for your great work!

Vrekrer, 2020-12-08:
Hi rselec

Thank you for remembering me about 4b0adc8
I did not merge that branch because I was thinking in putting all the documentation into a wiki.
But now you mention platform IO. Do platform IO reads the documentation from the headers? Is 4b0adc8 in the right format for automatic documentation?

About your tokens vs commands question, there are several ways to reuse tokens especially if you implement a lot of writes and queries for your commands. For example the `Ethernet_Instrument.ino` example uses 7 tokens for 7 commands even if "wasting" 3 tokens in `SYSTem:COMMunicate:LAN`

You should put `#define SCPI_MAX_TOKENS`  before `#include "Vrekrer_scpi_parser.h"`

Right now `SCPI_MAX_TOKENS` to the power of `SCPI_ARRAY_SYZE` should be smaller than `2^31`

rselec, 2020-12-09:
Hello Vrekrer,

thanks for the fast response.

No I don't think this is the right format, I would say better use comments according to the doxygen syntax. But to be honest I'm not a professional programmer and not very familiar with the automatic documentation stuff around doxygen and readthedocs. Anyhow some explanation somewhere is always better than nothing ;-)
Maybe you could come back to edmundsj for documentation, he sounds quiet experienced.

Ok, I don't thought about token reusage, so both cases are possible and you have to check according to your implementation of commands.

I tried to #define the things in the main before including the header, that results in the ugly warnings below:

`Linking .pio\build\sparkfun_promicro16\firmware.elf
.pio\libdeps\sparkfun_promicro16\Vrekrer SCPI parser\src/Vrekrer_scpi_parser.h:61:7: warning: type 'struct SCPI_Parser' violates one definition rule [-Wodr]
 class SCPI_Parser {
       ^
.pio\libdeps\sparkfun_promicro16\Vrekrer SCPI parser\src\Vrekrer_scpi_parser.h:61:7: note: a different type is defined in another translation unit
 class SCPI_Parser {
       ^
.pio\libdeps\sparkfun_promicro16\Vrekrer SCPI parser\src/Vrekrer_scpi_parser.h:79:32: note: the first difference of corresponding definitions is field 'tokens_'
   char *tokens_[SCPI_MAX_TOKENS];
                                ^
.pio\libdeps\sparkfun_promicro16\Vrekrer SCPI parser\src\Vrekrer_scpi_parser.h:79:32: note: a field of same name but different type is defined in another translation unit
   char *tokens_[SCPI_MAX_TOKENS];
Checking size .pio\build\sparkfun_promicro16\firmware.elf
Building .pio\build\sparkfun_promicro16\firmware.hex
                                ^
lto1.exe: note: array types have different bounds`

I don't think it had compiled always with the same definition of SCPI_MAX_TOKENS and SCPI_ARRAY_SYZE. Haven't checked function on real target yet.

Anyhow 2^31 should be enough, I would run out of memory much earlier ;-)

Thanks,
Robert

Paul202003, 2021-03-05:
Hi Vrekrer,

This is a great project!

Actually I came across the following 
If I change SCPI_MAX_COMMANDS (for example from 20 to 24) the SCPI-parser freezes.
Where is SCPI_MAX_COMMANDS monitored in the "Vrekrer_scpi_parser.cpp" file?
I use Arduino IDE 1.8.13 and Arduino MEGA board.

Please  let me know.
With kind regards,
Paul.

Vrekrer, 2021-03-10:
Hello
Looks like that is not possible to override define statements used by libraries. (using the Arduino IDE)

https://arduino.stackexchange.com/questions/20439/override-define-statements-in-libraries
https://www.reddit.com/r/arduino/comments/9dj3ql/overridingremoving_defines_from_a_lib/

you will need to modify the library header file.

I will try to find a workaround.

hb020, 2023-01-15:
Like [franciszekjuras](https://github.com/franciszekjuras), I changed the library to one header file, so I could change the relevant `#define`'s. Also modified the library:
- added "ignore" characters. Sometimes sender's line ending is `\n`, sometimes `\r\n`, so I just ignore `\r`.
- added more debug info in `PrintDebugInfo()`. The problem is that `RegisterCommand()` does not have proper protection against overflow of the pre-sized arrays (see `SCPI_ARRAY_SYZE`, `SCPI_MAX_TOKENS`, `SCPI_MAX_COMMANDS`). Not via return values, nor via debug logs. 

See my project [batterymeter](https://github.com/hb020/batterymeter/tree/master/batterymeter). Code is in the `batterymeter` directory (stupid Arduino IDE...)

palmerr23, 2023-11-15:
Silicon Chip Magazine project "Swiss Army Knife" [https://www.siliconchip.com.au/Issue/2023/April/Automated+Test+Bench](url)