Vrekrer/Vrekrer_scpi_parser #11: Make it available through Platform.io

vdesmedt, 2020-11-18:
I can create the necessary library.json and push it if you want me to.

Vrekrer, 2020-11-20:
Thanks.
library.json was created here: https://github.com/Vrekrer/Vrekrer_scpi_parser/pull/12

What else is needed?
Do I have to register the library in the Platform.io webpage? Where?

vdesmedt, 2020-11-20:
I noticed the author's URL is left blank which will be a problem to publish, either remove the tag of fill it.

To publish, using platform.io environment, use `pio package publish` like described here: https://docs.platformio.org/en/latest/core/userguide/package/cmd_publish.html#cmd-package-publish

I can of course do it for you if you wish

Vrekrer, 2020-11-20:
I have been reading the Platform.io [docs](https://docs.platformio.org/en/latest/librarymanager/creating.html)
And we can use library.properties to publish the library.
So, I have reverted #12 

I don't want to create a Platform.io account, please publish it yourself.
If I am not wrong, you need to run
`pio package publish  https://raw.githubusercontent.com/Vrekrer/Vrekrer_scpi_parser/master/library.properties`

vdesmedt, 2020-11-21:
https://platformio.org/lib/show/11457/Vrekrer%20SCPI%20parser

Apparently, somehow, you just did ;)

closed 2023-02-26 by Vrekrer