Vrekrer/Vrekrer_scpi_parser #24: Robustness

Volhout, 2023-01-06:
Dear Vrekrer,

Just tried your parser (UNO). Works fine on Arduino Monitor and when I send compact strings (i.e. echo "string" > /dev/ttyUSB0).
However it seems not to handle other terminal programs (Putty / Teraterm) that send characters directly when moment typed.

Any fix for that ?

Thanks

bogus43, 2023-01-07:
Hello @Volhout,
It might be helpful, if You change 2 putty settings: navigate to tab "Terminal" and check "Local echo" -> Force on and "Local line editor" -> Force on. If I am correct, putty send '\r' by default, so it should be also Your termination character in Arduino code. I no longer use putty, maybe default termination character could be changed.

Vrekrer, 2023-02-13:
Thanks bogus43.
Yes, this seems to be a termination chars problem.
We use line feed `\n` as termination char for all the examples.

closed 2023-02-13 by Vrekrer