Vrekrer/Vrekrer_scpi_parser #25: Communication issue between Python and Arduino

LeBrownBrun, 2023-01-26:
Dear Vrekrer,

I really want to use your scpi parser library but I am having issues communicating between arduino and my python program. 

E.g. I am using your Command_Tree_Check.ino script to check if the message "*IDN?" is coming through. If I open the serial monitor and send "*IDN?" it returns - as it should - "Vrekrer,Hash check example,#00,v0.4.2". So the Identify() works.

My python code looks like this:

`
import serial
import time

ser = serial.Serial("COM3", 9600)
time.sleep(3)
print("port was opened.")

// to read Debug Info
i = 0
while True:  
    text = ser.read_until()
    print(text)
    i += 1
    if i == 23:
        break

while True:
    input_value = input("Press enter to send.")
    ser.write("*IDN?".encode())
    time.sleep(1)
    print(ser.readline())`

After I press enter to send the "*IDN?" command the serial just gets stuck without any return.  

Since the "*IDN?" command gets "accepted" over the arduino serial monitor but somehow not over the serial communication from my python code, I assume there must be some decoding issues. But I do not know how to fix that. Also if I try with a serial communication program like HTerm I can connect with the arduino but same here, I do not get any return if I send "*IDN?". 

A simple serial communication between arduino and python works fine (shown below). I can let the LED blink when sending "A" and it returns "Message received" each time I hit enter. 

**Arduino code:**
`char stat;

void setup() {
  Serial.begin(9600); 
  pinMode(LED_BUILTIN, OUTPUT);
}
void loop() {
  // check for the serial data at serial port
  if (Serial.available()) {
    // read byte of received data:
    stat = Serial.read();
    if (stat == 'A') {
      Serial.println("Message recieved");
      digitalWrite(LED_BUILTIN, HIGH);
      delay(1000);
      digitalWrite(LED_BUILTIN, LOW);
      delay(1000);
      digitalWrite(LED_BUILTIN, HIGH);
      delay(1000);
      digitalWrite(LED_BUILTIN, LOW);
      delay(1000);
    }
}`

**Python Code:**
`
import serial
import time

ser = serial.Serial("COM3", 9600)
time.sleep(3)
print("port was opend.")

while True:
    input_value = input("Press enter to send.")
    ser.write("A".encode())
    time.sleep(1)
    print(ser.readline())`

Very thankful for any help!
Best,
LeBrownBrun

Vrekrer, 2023-02-13:
Hi LeBrownBrun

You need to send a termination char `\n` in your python script
`ser.write("*IDN?\n".encode())`

Instead of using the `serial` lib, try using the `pyvisa` lib
```
import pyvisa
rm = pyvisa.ResourceManager('@py')
# use rm.list_resources() to get the resource names
myInstrument = rm.get_instrument('COM3::INSTR')
myInstrument.write_termination = '\n'
myInstrument.read_termination = '\n'
print( myInstrument.query('*IDN?') )
```

LeBrownBrun, 2023-02-22:
Thank you for your reply.
I started with ```pyvisa``` lib but since it gave me always a timeout error, I wanted to identify the problem and therefore used the ```serial``` lib. 

Adding the termination char ```\n``` works great now with the ```serial``` lib. However, when I try to switch back to ```pyvisa``` lib, I am still getting timeout errors with the following code (btw. I think the command get_instrument is not used anymore in the new versions. Also, I installed NI backends and therefore, not using the @py backend):
```
import pyvisa
import time

rm = pyvisa.ResourceManager()
print(rm.list_resources())
myInstrument = rm.open_resource('ASRL3::INSTR')
time.sleep(3) #to give the arduino time to restart
print("Opened successfully.")
myInstrument.write_termination = '\n'
myInstrument.read_termination = '\n'
print( myInstrument.query('*IDN?') )
# myInstrument.write('*IDN?') 
# print(myInstrument.read())

```
error message:
```
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

On the other hand, when I am using another scpi parser like ```<scpiparser.h>``` it works fine with the ```pyvisa``` lib using the same code as above. But I want to use your parser as it is more convenient to use and gives a better general code overview.
Maybe you find some time to figure out where the issue is.

Thank you!

Vrekrer, 2023-02-23:
Maybe is the configuration of the serial port.

Try this:

```Python
from pyvisa.constants import StopBits, Parity
rm=pyvisa.ResourceManager()
myInstrument = rm.open_resource('ASRL3::INSTR', 
    baud_rate=9600, data_bits=8, flow_control=0, 
    parity=Parity.none, stop_bits=StopBits.one)
```

LeBrownBrun, 2023-02-23:
Unfortunately, this did not help. 
But I know now where the issue is. I noticed that the response from your parser was slightly slower than the one from the other parser. So I increased the ````timeout```` to 5s. Now it works fine.

This is my code now:
````
import pyvisa
import time

rm = pyvisa.ResourceManager()
print(rm.list_resources())
myInstrument = rm.open_resource('ASRL3::INSTR', 
    write_termination = '\n', read_termination = '\n', 
    timeout=5000)
time.sleep(3) #to give the arduino time to restart
print("Opened successfully.")
print(myInstrument.query('*IDN?'))
````

Thanks for your help!

closed 2023-02-26 by Vrekrer