Vrekrer/Vrekrer_scpi_parser #27: Compile error while using library outside Arduino/libraries

bogus43, 2023-02-14:
Hi,
I guess my problem is a little trivial for someone here, but I have trouble using the latest version of library. It works fine when library is in Arduino/library directory, but in my project I have all libraries in sub-directory in my project. And I cannot compile that - error message: "Vrekrer_scpi_parser.h:196:10: fatal error: Vrekrer_scpi_parser.hc: No such file or directory"
Do someone have some suggestions?

Vrekrer, 2023-02-14:
3b835a should fix that.
(Rename Vrekrer_scpi_parser.hc to Vrekrer_scpi_parser_code.h)

bogus43, 2023-02-14:
Now it works fine, thank You so much.

bogus43, 2023-02-20:
This solution works fine only if library is included once in the main Arduino file. If I want to include more  (I have more dependent files), there is linker error - multiple definition of... . To be honest, it is not a problem for me - I renamed a header files to .cpp and include Vrekrer_scpi_parser.h to this files (It is ok for me if I cannot change number of tokens, I can leave it as constant value). But maybe there is a simple solution? Below I attached test code.

```
// Arduino INO file
#include "src/Vrekrer_scpi_parser.h"
#include "Data.h"

SCPI_Parser Arduino;

void setup()
{
    Serial.begin(9600);
    Arduino.RegisterCommand("TEST", &Test1);
}

void loop()
{
    Arduino.ProcessInput(Serial, "\n");
}
```
```
// Data.h 
#ifndef _DATA_FILE_H_
#define _DATA_FILE_H_

#include "src/Vrekrer_scpi_parser.h"

void Test1(SCPI_C commands, SCPI_P parameters, Stream &interface);

#endif
```
```
//Data.cpp
#include "Data.h"

void Test1(SCPI_C commands, SCPI_P parameters, Stream &interface)
{
    interface.println("Test 1");
}
```
**SRC** directory contain library files.

Vrekrer, 2023-02-20:
Unluckily, that is a problem with the Arduino compiler.
Including all the .cpp code inside the .h file should work.

```
// Arduino INO file
#include "src/Vrekrer_scpi_parser.h"
#include "Data.h"

SCPI_Parser Arduino;

void setup()
{
    Serial.begin(9600);
    Arduino.RegisterCommand("TEST", &Test1);
}

void loop()
{
    Arduino.ProcessInput(Serial, "\n");
}
```

```
// Data.h 
#include "src/Vrekrer_scpi_parser.h"

void Test1(SCPI_C commands, SCPI_P parameters, Stream &interface)
{
    interface.println("Test 1");
}
```

```
// Data.cpp
//Not needed, or empty file
```

closed 2023-02-26 by Vrekrer