Vrekrer/Vrekrer_scpi_parser #31: Suggest to disable DTR setting

caizikun, 2023-10-14:
In some Serial Port communication software we need to enable DTR setting for normal working.  I use Arduino IDE ,RP2040 and earlephilhower/arduino-pico 
pls ref to :
https://github.com/hathach/tinyusb/pull/401
https://github.com/hathach/tinyusb/issues/482
https://github.com/raspberrypi/pico-sdk/issues/906
https://github.com/raspberrypi/pico-sdk/pull/932
https://github.com/hathach/tinyusb/pull/506
https://github.com/earlephilhower/arduino-pico/issues/1620
https://github.com/earlephilhower/arduino-pico/pull/1624
https://github.com/earlephilhower/arduino-pico/issues/1678
https://forums.raspberrypi.com/viewtopic.php?t=335296

Now I add **Serial.ignoreFlowControl();**  in void setup() to ignore DTR setting. 

void setup()
{
    my_instrument.RegisterCommand(F("*IDN?"), &Identify);
  Serial.begin(115200);
Serial.ignoreFlowControl(true);
}

Vrekrer, 2023-10-14:
Hello.
Vrekrer_scpi_parser is only a parser for the SCPI protocol.
Vrekrer_scpi_parser only process text (usually obtained from a communication hardware)

If this is a communication problem, please report it to the library that enable your communication hardware.

Vrekrer, 2023-10-14:
In `Vrekrer_scpi_parser` you set up your communication device, and then pass it to the library as a stream.

Your `Serial.ignoreFlowControl()` "solution" seems to be the right way to solve your "problem".
You are just setting up your `Serial` hardware.

closed 2023-10-14 by Vrekrer