Vrekrer/Vrekrer_scpi_parser #32: nummeric suffix in SetCommandTreeBase

nerdyscout, 2023-10-16:
Hi @Vrekrer

according to your  Readme suffixes could be in a SetCommandTreeBase... atleast I do understand the part  **"CHANnel#:** so.

```
Numeric suffixes using the # character:
E.g. definition : "CHANnel#:SELect"
E.g. usage : "CHAN0:SEL", "chan5:sel", "chan13:sel"
```
The Numeric_suffixes.ino example uses suffixes only in a command, which works fine. 

I tried to build something like this by myself
```
  scpi.SetCommandTreeBase("DUT#");
  scpi.RegisterCommand(":STATe", &setDUTState);
  scpi.RegisterCommand(":STATe?", &getDUTState);
```
but commands.First() inside the setDUTState function returns just some strange "�Y"

are numeric suffixes in ComanndTrees supposed to work?
are you willing to update the example could to show how it is done?

thanks!

Vrekrer, 2023-10-16:
Hi @nerdyscout 

That might be a bug.
I will have time to look at it only next week.
Thanks for the report.

nerdyscout, 2023-10-16:
turns out it works fine! thank you for your quick reply.
but maybe still a good idea to adopt the suffix example.

closed 2023-10-16 by nerdyscout