Vrekrer/Vrekrer_scpi_parser #35: CRLF not recognised as end of command string.

PE1RCD, 2024-10-05:
I am working on a CW generator  and implemented this  parser. 
Sending commands from the arduino IDE works fine. 

Communication from PUTTY or any other terminal program wil send no reply after sending a command. 

I can reproduce this behaviour when i set the terminal from the IDE to also send CR before the LF.

Vrekrer, 2024-10-05:
Hi,

Have you set the appropriate end characters in your software?

For example:
`my_instrument.ProcessInput(Serial, "\n");`  This is used in the examples and expects an LF at the end of your messages
`my_instrument.ProcessInput(Serial, "\r\n");`  Use this if you want to use CR+LF as the end characters for your messages

I realize now that this is not well explained in the examples.
Thank you for the "bug" report.

PE1RCD, 2024-10-05:
Will try this evening. 

I looked into the documentation, but couldn't find a 100% answer.

Thanks for the quick reply.

PE1RCD, 2024-10-05:
Added  _my_instrument.ProcessInput(Serial, "\r\n");_to my setup.

PUTTY and my other software still don't get a replay after sending the command. \

`void setup(){
  
 
     dds.begin();
    my_instrument.ProcessInput(Serial, "\r\n");
  
  
  my_instrument.RegisterCommand(F("*IDN?"), &Identify);
  
    my_instrument.SetCommandTreeBase(F("SOURce:"));
    my_instrument.RegisterCommand(F(":FREQuency#"), &SetFrequency);
    my_instrument.RegisterCommand(F(":FREQuency#?"), &GetFrequency);
    my_instrument.RegisterCommand(F(":SELEct"), &SetFSEL);
    my_instrument.RegisterCommand(F(":SELEct?"), &GetFSEL);
    my_instrument.RegisterCommand(F(":ENABle"), &SetEnable);
    my_instrument.RegisterCommand(F(":ENABle?"), &GetEnable);

 
  Serial.begin(9600);
  Serial.println("PE1RCD,SCPI CW_GEN,#01");
  
 
  dds.setFrequencyHz(0, F[0]);
  dds.setFrequencyHz(1, F[1]);
  dds.selectFrequencyRegister(0);

    dds.setPhaseDeg(0,0);
    dds.selectPhaseRegister(0);
//dds.enable();

 }`

no clue what i am doing wrong.

Vrekrer, 2024-10-05:
It looks like PuTTY sends each character you type to the serial interface.
Try setting the timeout variable to a large value:
 `my_instrument.timeout = 60000  //one minute`
See the [error handling example](https://github.com/Vrekrer/Vrekrer_scpi_parser/blob/master/examples/Error_Handling/Error_Handling.ino) for further details.

You should also verify your serial setup, including termination characters, and other parameters such as baud rate, stop bits, parity, etc.
https://pbxbook.com/voip/sputty.html
https://stackoverflow.com/questions/11997108/what-does-putty-send-when-i-press-enter-key

Also try the [RawData_Parameters example](https://github.com/Vrekrer/Vrekrer_scpi_parser/blob/master/examples/RawData_Parameters/RawData_Parameters.ino). 
Sending `ECHO:SPEcial <-a_space_is_needed_here` should send you back all your transmission including the end characters.

Does it work with CR + LF using the Arduino IDE?
If it works, then the problem is not with the parser.

PE1RCD, 2024-10-09:
> It looks like PuTTY sends each character you type to the serial interface. 

You are correct.  Verified with logic analyzer
[image]

arduino IDE 
[image]

Try setting the timeout variable to a large value: `my_instrument.timeout = 60000 //one minute` 
>
> 
> Does it work with CR + LF using the Arduino IDE? If it works, then the problem is not with the parser.
It looks like /n/r or /n /r doesn't appear to change the behaviour

will do some more tests later this week.