Vrekrer/Vrekrer_scpi_parser #37: overflow errors

Steffen-W, 2025-01-16:
Hi @Vrekrer, 
i have been searching for a very long time for an error that was due to an overflow. 
Can you perhaps issue a message when an overflow occurs? That would be extremely useful.

Steffen-W, 2025-01-16:
It would at least be extremely helpful to have a function that can be used to check whether one of the following variables is true.

```cpp
  struct internal_errors {
    //Command storage overflow error
    bool command_overflow = false;
    //Token storage overflow error
    bool token_overflow = false;
    //Branch (SCPI_Commands) storage overflow error
    bool branch_overflow = false;
    //Special command storage overflow error
    bool special_command_overflow = false;
  } setup_errors;
```

Vrekrer, 2025-01-16:
Hi Steffen

Have you tried the [`error handling example`](https://github.com/Vrekrer/Vrekrer_scpi_parser/blob/master/examples/Error_Handling/Error_Handling.ino)?

Steffen-W, 2025-01-16:
I think your project is great. No, I somehow overlooked it. Many thanks for the quick reply. That helps me.

Vrekrer, 2025-01-16:
Hi.

You can increase the number of commands.
Look at the [Configuration options example.](https://github.com/Vrekrer/Vrekrer_scpi_parser/blob/master/examples/Configuration_Options/Configuration_Options.ino)

The default number of commands is  for low-RAM processors, such as Arduino.

Steffen-W, 2025-01-16:
Hi,

I did that but only after I had found the problem. That's why I would have liked it if I could have simply called up the overflow in a function. According to the motto there is an error or not.

Vrekrer, 2025-01-16:
That is a good idea. 
I will add an status to the library class and try to report class errors there.

Steffen-W, 2025-01-16:
Something along the lines of buffer_ok() which then outputs true if there is no overflow, you could very well check this once after initialization, then you won't be surprised if there are errors.