Vrekrer/Vrekrer_scpi_parser #39: Linking issue / Multiple definitions

aanas-sayed, 2025-04-19:
### Issue Summary

I'm getting a linker error when splitting my SCPI command setup into separate files:

```
C:\Users\xxx\AppData\Local\Temp\arduino_build_706583\sketch\scpi_interface.cpp.o (symbol from plugin): In function `DefaultErrorHandler(SCPI_Commands, SCPI_Parameters, Stream&)':
(.text+0x0): multiple definition of `DefaultErrorHandler(SCPI_Commands, SCPI_Parameters, Stream&)'
```

### Works:

- Everything compiles fine if I put all SCPI setup and handlers in `main.ino`.

### Fails:

- When I move SCPI setup into `scpi_implementation.h` (included only in `main.ino`), I get the error above.

### Details:

- `Vrekrer_scpi_parser.h` is only included in `scpi_implementation.h`, not multiple times.
- The issue seems to come from this in the header:

```cpp
#include "Vrekrer_scpi_parser_code.h"
```

Since that file includes implementations, including it from a `.h` file causes multiple definitions when the header is used across files.

### Workaround:

- Putting everything in `main.ino` avoids the issue, probably because it's compiled as one translation unit.

Thanks so much for this great project — I was using another SCPI library before, but it was too heavy for my needs. This one is much more lightweight and exactly what I was looking for!

Would love to know if there’s a recommended way to structure this, or if you’ve considered moving the implementations to `.cpp` files to avoid these issues. I understand there’s likely a reason for the current setup — I just can’t quite follow the rationale yet.

aanas-sayed, 2025-04-19:
Btw I saw the note now to try:

```c
#define VREKRER_SCPI_PARSER_NO_IMPL
#include "Vrekrer_scpi_parser.h"
```

But I still get the issue. I put the above code in my `scpi_helper.h` file and then in my `main.ino` I used only `#include "scpi_interface.h"`

Am I doing something wrong?

Vrekrer, 2025-04-20:
Hello.

The Arduino compiler can be quite complex, and I don't really understand it.
However, I think that including a `.h` file (without a corresponding `.cpp` file) is essentially equivalent to copying and pasting its contents at the location of the `#include` directive.

If you already have a working `main.ino` file, you can simply move chunks of code into separate files.
Just ensure that you manage the scope correctly and that all functions, variables, etc are declared before they are used—similar to an standard C compiler.

Hope that helps.

aanas-sayed, 2025-04-20:
Hi, I've tried what you have suggested but it still has issues. Even if the code completely works fine in `main.ino` file, moving it into any seperate file causes linking issues. I've tried all combinations - only using a `.h` file, only a `.cpp` file and also both.

I've dived deeper into the issue and its seems to be linked to having the implementation code in .h files. I can see the reasoning was so that you can have users define variables and they can change the max token etc. But after running some tests I seem to be able to achieve this whilst keeping the files seperate. After doing so I was able to solve the issue and it works now regardless of where you import it. 

I've made a fork and pushed the code. I'll make a PR and hope you can review it and see if you agree with it. I've also added an option for a user settable custom error code enum. 😄 

Thank you!